Zanduino/MCP7940 #5: No need to wait after Wire.requestFrom()

Koepel, 2017-08-13:
In the file "MCP7940.cpp" there are while-loops with a timeout after the Wire.requestFrom. Those while-loops and those timeouts are not needed and can be removed.

When the Wire.requestFrom() returns, the I2C transaction has completely finished and the received data is waiting in a buffer in the Wire library.

To check if the I2C transaction was successful, you may compare the number of requested bytes to the number of bytes that have been received and are waiting in a buffer.

```
Wire.requestFrom( Wire.requestFrom(MCP7940_ADDRESS, (uint8_t)7);
if( Wire.available() == 7) {
  // good
} else {
  // fail
}
```

SV-Zanshin, 2017-08-13:
Thanks - the code is indeed extraneous and has been removed.

closed 2017-08-13 by SV-Zanshin

Koepel, 2017-08-13:
When you request 4 or 7 bytes and that fails, you still call the function DateTime with parameters that are not valid.

It probably doesn't matter, since the I2C bus should be reliable or else the hardware is wrong. In case the hardware is unreliable or the chip is not connected or with a collision in a multi-master bus, the Wire.available() could return zero. When Wire.available() is zero, a call to Wire.read() returns -1.

SV-Zanshin, 2017-08-13:
I looked into that and since something needs to be returned in case of an I2C error using 0 values for all the parameters to DateTime() is as close as I can get.

Koepel, 2017-08-13:
I see, cool.