Zanduino/MCP7940 #6: ESP8266 (Generic Module)

Tensa225, 2017-10-31:
MCP7940-master/MCP7940.h:176:19: error: extra qualification 'MCP7940_Class::' on member 'readRAM' [-fpermissive]

         uint8_t&  MCP7940_Class::readRAM(const uint8_t addr,T &value) {       //                                  //

                   ^

MCP7940-master/MCP7940.h:190:12: error: extra qualification 'MCP7940_Class::' on member 'writeRAM' [-fpermissive]

       bool MCP7940_Class::writeRAM(const uint8_t addr, const T &value) {      // the MCP7940 SRAM                 //

            ^

exit status 1
Error compiling for board Generic ESP8266 Module.

SV-Zanshin, 2017-10-31:
I opened up the example "AccessMemory.ino" (using  the standard 1.8.3 IDE and also with Atmel Studio with VisualMicro installed) with the "Arduino Esplora" selected and got a good compile:

`Compiling 'AccessMemory' for 'Arduino Esplora'
Program size: 9,876 bytes (used 34% of a 28,672 byte maximum) (16.17 secs)
Minimum Memory Usage: 477 bytes (19% of a 2560 byte maximum)
 `

Could you try to compile that example to see if you also get an error? Also, I don't have an "Esplora8266" in my list of boards.  Could you try to set the compiler options to "verbose" and see if the error message is more detailed?

SV-Zanshin, 2017-10-31:
Ok, it seems that the template functions "readRAM" and "writeRAM" do not need the "MCP7940_Class::" definitions. I don't know why your compiler doesn't like them, but they are superfluous and I've removed them from the header file.

SV-Zanshin, 2017-10-31:
Created Release 1.0.5a for this fix.

closed 2017-10-31 by SV-Zanshin

Tensa225, 2017-11-02:
Thanks for the quick response.