Zanduino/MCP7940 #7: SAMD21 board support

newerbee, 2017-11-11:
Hi Zanshin,

Nice job with the library. But It looks like the library doesn't compile for SAMD21 based boards such as Arduino M0, or adafruit Feather M0. I am not an expert, but I happen to be using these boards...

BR.
Newerbee

SV-Zanshin, 2017-12-18:
Let me look into that. I don't have any of those boards, but can try to set my IDE to one of them and see what the problem might be.

SV-Zanshin, 2018-04-29:
Sorry that this took so long to look into!

The reason was that the compiler settings were different and I was setting parameter defaults in the functions as well as in the prototype definitions. I've removed those from the function definitions and now the library compiles without a problem on SAMD21 boards.

closed 2018-04-29 by SV-Zanshin