Zanduino/MCP7940 #11: Alarm problem #2

logicaprogrammabile, 2018-04-29:
Hi 
i have tested the new version but i have another issue. if i run the code:
`#include <MCP7940.h>

MCP7940_Class MCP7940;
DateTime now;
DateTime allarme;

enum alarmTypes { matchSeconds, matchMinutes, matchHours, matchDayOfWeek,     // Enumeration of alarm types       //
                  matchDayOfMonth, Unused1, Unused2, matchAll, Unknown
                };

void setup() {
  // put your setup code here, to run once:
  Serial.begin(9600);

  MCP7940.adjust(DateTime(2018, 04, 29, 23, 59, 45));
  MCP7940.setAlarm(1, 7, DateTime(2018, 04, 30, 0, 0, 10));

}

void loop() {
  // put your main code here, to run repeatedly:
  delay(1000);
  now = MCP7940.now();
  Serial.print(now.day());
  Serial.print("/");
  Serial.print(now.month());
  Serial.print("/");
  Serial.print(now.year());
  Serial.print(" ");
  Serial.print(now.hour());
  Serial.print(":");
  Serial.print(now.minute());
  Serial.print(":");
  Serial.println(now.second());

  if (MCP7940.isAlarm(0)) {
    Serial.print("Alarm0*");
    MCP7940.clearAlarm(0);
  }`

the alarm is generated at 2018/04//30 0:0:0 but i expected the alarm at 2018/04//30 0:0:10

regards
Marco Lai

SV-Zanshin, 2018-04-29:
Hello Marco,

I found a MCP7940 and oscillator and hooked it up to test the problem.

The alarm is being set correctly; in your code you write to ALARM1 but check ALARM0  :)

closed 2018-04-29 by SV-Zanshin

logicaprogrammabile, 2018-04-29:
oh oh!! typical mistake of the copy/paste.
i retest the code and all run correctly.
sorry for my error!!
thank you

Marco