Zanduino/MCP7940 #12: Battery backup

logicaprogrammabile, 2018-05-21:
Hello
I have a problem. after setting the time and date in the mcp7940, if I remove main power to the RTC and after a few minutes, i give power to the RTC, the time and date are reset to 1/1/2001 0:0:0.
(the battery backup  is connected and it is charged)
the time data is not retained in memory and is not increased!!

you can verify this

regards

SV-Zanshin, 2018-05-22:
Did you issue a call to "setBattery(true);" in your code? The default setting for the MCP7940 is that the battery backup is disabled.

logicaprogrammabile, 2018-05-22:
no! I did not set setBattery (true); I did not notice this feature in the library !! I'll do a test tonight. thank you

SV-Zanshin, 2018-05-22:
No worries - I'm glad that the library is working for you!

closed 2018-05-22 by SV-Zanshin