Zanduino/MCP7940 #26: Bit Mask hour register

logicaprogrammabile, 2018-07-08:
hi
in your code the mask bit for the hour register is 7F

from DateTime MCP7940_Class::now()

`_hh = bcd2int(Wire.read() & 0x7F);`

the correct mask should be 0x3F because the bit6 set the 12/24 format and should be exclude it 

the mask bit is present in the follow functions
DateTime MCP7940_Class::now()
DateTime MCP7940_Class::getPowerDown()
DateTime MCP7940_Class::getPowerUp() 

regards
Marco

SV-Zanshin, 2018-07-08:
Good catch on the mask.  The library currently works in 24-Hour mode exclusively, so missing the bit that represents the AM/PM when set to 1 wasn't noticed during testing; but it is better to fix the error should the library be expanded to support AM/PM 12-hour mode in the future.

SV-Zanshin, 2018-07-08:
Added fix to master branch

closed 2018-07-08 by SV-Zanshin