Zanduino/MCP7940 #28: Autocalibration

logicaprogrammabile, 2018-07-25:
Hi, I saw that you created the Square Wave Output function. The mcp7940 has two methods to be calibrated. One is what you have already implemented. the second is to measure the quartz frequency and correct the difference between the measurement and the nominal value.
Could you create a function that allows you to measure the frequency and calibrate the mcp7940?

rif datasheet equation 5-2 (calculating trim value from measured frequency)

SV-Zanshin, 2018-07-27:
The Datasheet contains a formula which will allow the calibration to be set when the actual frequency is known. I don't have an oscilliscope available to test any code, but can add this to library and either trust that it works or have it checked by someone with the right Hardware.

logicaprogrammabile, 2018-07-27:
ok i have an oscilloscope and a frequency meter and i can try to test the function

logicaprogrammabile, 2018-07-27:
for the autocalibration the arduino/mcu should be able to read the frequency and the difference between the nominal frequency (32768 hz) , calculated using the formula, this data will be use to calibrate the rtc

SV-Zanshin, 2018-07-27:
The Problem is that the crystals used for UNOs etc. are less accurate than the crystals used for the RTCs, so using that as a source for an automated calibration of the RTC doesn't make too much sense.

logicaprogrammabile, 2018-07-27:
certainly, but it is the only way to do an auto calibration. I have several systems that use the rtc7940 and I must say that after some time I have a significant time deviation. it is not possible to calibrate each time. even if with some errors it would be important that arduino perform the calibration

SV-Zanshin, 2018-07-28:
I've got an example sketch put together to do the calibration using the Arduino CPU speed and a hardware interrupt function, but I've got some problems in achieving a viable calibration. I'll continue experimenting and once I have it working I'll upload the calibration example and see if that meets what you are looking for.

SV-Zanshin, 2018-08-04:
I've added an overloaded definition for the calibrate() function. If you call it with a floating-point parameter then it will consider that to be a Hz value that has been measured. The function will retrieve the current Square-Wave Hz and apply the calibration formula in the documentation to generate a trim offset.

I cannot get accurate readings from my current Arduino setup with which to write an automatic calibration program, the values fluctuate with higher values than the code can compensate for.

SV-Zanshin, 2018-08-04:
I have a test program but have determined that my Arduino is so far off in terms of timing that the fine-trim doesn't suffice and I'd need to use coarse mode and then the MCP7940 clock is completely wrong.

I'll leave the example program at https://github.com/SV-Zanshin/MCP7940/blob/master/examples/ArduinoClockCalibrate/ArduinoClockCalibrate.ino up for while,  but I'll be deleting it sometime as it doesn't really work correctly.

closed 2018-08-04 by SV-Zanshin

logicaprogrammabile, 2018-08-04:
ok thanks, I will perform some tests using the oscilloscope