Zanduino/MCP7940 #33: MCP7940.adjust() freezing in main loop

stuveri, 2018-08-30:
Hello,

I'm finding that adjust() is working ok on initialisation in the setup() routine
however I have a menu-driven date and time input system to allow the user to 
set their correct time. When I just use setTime() from the Arduino library it works
(at least sets the time locally) however if I use the the MCP7940.adjust(DateTime)
the system freezes. At the end of the menu input I have even hardcoded a value
to see if I can get it to work: MCP7940.adjust(DateTime(2020,9,9,18,19,20));
but that has the same problem.

It is strange that during setup I can adjust the date to any hardcoded value using on the
RTC but this doesn't help me allow the user to set their own date/time.

Any ideas would be appreciated,
thanks,
Stuart

SV-Zanshin, 2018-08-30:
Let me dig up a MCP7940 to see if I can reproduce the problem. Which MCP are you using and on which Arduino processor (perhaps that might play a role in causing the problem).

stuveri, 2018-08-30:
Hi,

I'm using an Arduino Micro so 32U4 processor and an MCP7940N RTC.
My application also uses interrupts in the main loop to read a data line
and I was wondering if that might be part of the problem.

Thanks for having a look at this.

Stuart

SV-Zanshin, 2018-08-31:
I've got both of those (the Micro and MCP7940N) and will check it tonight. Any time when one programs with interrupts and experiences "hangs" there is a good chance that it is due to that. You might try to use the LED to show that the interrupt is being fired "(digital.write(!digital.read(13));" or use the LED in the MCP7940.adjust() code. 
I do seem to remember that I had some issues with the adjust() but can't remember if it was hanging or invalid values.

stuveri, 2018-08-31:
Hi,
Before you spend any more time, I think the freezing problem is solved.
Although I claimed the adjust() was being called in the main loop, it was actually being called
at the end of an ISR used to detect the last button press of a time/date set menu function. 
Of course an ISR isn't the place for this call, and it now seems to work using a flag so that
the function invoked back the main loop to set the time/date on the 7940.
Will do some more testing but I think it's now ok.
thanks for your help.

SV-Zanshin, 2018-08-31:
I've got a non-Micro Arduino setup with the MCP7940N and used multiple calls to adjust() with no problems, so I think that you've most likely found the problem. I'll close this issue but don't hesitate to open a new one if necessary.

closed 2018-08-31 by SV-Zanshin