Zanduino/MCP7940 #34: 'BUFFER_LENGTH' was not declared in this scope

LongmontRobotics, 2018-09-23:
In file included from C:\Users\Donald\Documents\Arduino\libraries\MCP7940\examples\SquareWave\SquareWave.ino:37:0:

C:\Users\Donald\Documents\Arduino\libraries\MCP7940\src/MCP7940.h: In member function 'uint8_t& MCP7940_Class::readRAM(uint8_t, T&)':

C:\Users\Donald\Documents\Arduino\libraries\MCP7940\src/MCP7940.h:238:17: error: 'BUFFER_LENGTH' was not declared in this scope

           if (i%BUFFER_LENGTH==0) {                                           // Read I2C again on buffer boundary//

SV-Zanshin, 2018-09-24:
I just tried to compile for an ESP32 (which I happened to purchase this weekend) and get the compiler error you mentioned. 
Unfortunately when I fixed that error, I received an error regarding the interrupt handling, which is done differently on the ESP32 than with Atmel chips.  I will fix this for the ESP32 but don't know if that is the microprocessor you are using.

SV-Zanshin, 2018-09-24:
I've uploaded a new version of the SquareWave.ino program which should fix the interrupt issue.

closed 2018-09-24 by SV-Zanshin

SV-Zanshin, 2025-02-06:
@typetrade - I'm not sure what your message means, it looks like you've pasted the source code from the current header file. But there's no comment and this issue was closed in 2018...