Zanduino/MCP7940 #36: I2C clock config uses uint16_t values

hexeguitar, 2018-12-14:
The constants and the function arguments for setting the I2C clock speed are uint_16t, but the actual values are out of range (100000, 400000), hence they are truncated (compiler throws a warning).

Fix:
in MCP7940.h line 72 & 73
```c
#ifndef I2C_MODES                                                           // I2C related constants            //
  #define I2C_MODES                                                         // Guard code to prevent multiple   //
  const uint16_t I2C_STANDARD_MODE      =     100000;                       // Default normal I2C 100KHz speed  //
  const uint16_t I2C_FAST_MODE          =     400000;                       // Fast mode                        //
#endif  
```
line 185:
```c
      bool     begin(const uint16_t i2cSpeed = I2C_STANDARD_MODE); 
```
in MCP7940.cpp line 231
```c
bool MCP7940_Class::begin(const uint16_t i2cSpeed) {
...
```
change the uint16_t to uint32_t
Arduino Wire library expects an uint32_t passed to the setClock method.

SV-Zanshin, 2018-12-15:
Good catch - added fix
Thanks!

closed 2018-12-15 by SV-Zanshin

davidlehrian, 2019-01-26:
I just updated my library and got the fix that hexeguitar mentioned above, and while it corrected the truncation, I now get a warning, "MCP7940.h: 68:49: warning: large integer implicitly truncated to unsigned type" for both lines 68 and 69.  The integers 100000 and 400000 need to be case to (uint16_t) as:

`    const uint16_t I2C_STANDARD_MODE      =     (uint16_t)100000;                       // Default normal I2C 100KHz speed  //
    const uint16_t I2C_FAST_MODE          =     (uint16_t)400000;                       // Fast mode                        //
`
and the warning goes away.

davidlehrian, 2019-01-27:
Thanks for the reply.  I should have caught that when I read this thread.  The code he quoted was the code that he changed FROM not what he changed the code TO.  And if I thought about it I should have realized that 16 bits only goes up to 65536.  Anyway, apparently the version of the library that gets installed via the Arduino interface has this error in it.  I don't run the Arduino interface much but I updated it the other day as a new project required an updated version and I also updated all my installed libraries.  The MCP7940 library was updated to Version 1.1.1 and when I compiled my project I saw this error.  I don't know how the code gets from GitHub to the Arduino Library Manager, but it needs to be updated.

SV-Zanshin, 2019-01-27:
The most recent version of the library is 1.1.5 and I don't know why the library manager isn't picking up the newest (and most correct) version.

I've opend up a ticket for this at https://github.com/arduino/Arduino/issues/8450 and we'll see what might be causing the issue. If you don't want to wait for the correct version to show up in the IDE's library manager you could always download the most recent version and install it manually.

davidlehrian, 2019-01-27:
Ah, interesting.  

No worries, I fixed the warning in my version of the library.  I'll watch for Version 1.1.5.

davidlehrian, 2019-01-28:
I just updated via the Arduino Library Manager and it indeed updated to version 1.1.6 and compiles w/o warning.  Thanks for looking at this.