Zanduino/MCP7940 #37: setAlarm overwrites AlamPolarity bit

INemesisI, 2019-01-17:
If setAlarmPolarity(1) is called before setting an alarm through setAlarm() the polarity will be forced to zero again.
I don't think this is intended, since changing the polarity after setting up the alarm can cause the MFP to have the wrong polarity for the first few us.

https://github.com/SV-Zanshin/MCP7940/blob/0678a66569f9ea81a9d959c8c6cce5247a6916eb/src/MCP7940.cpp#L592

Fix: also keep the polarity bit. I also force the dow to be only 3 bit long to prevent an accidental overwrite.
``` c++
    uint8_t wkdayRegister = readByte(MCP7940_ALM0WKDAY + offset);             //                                  //
    wkdayRegister &= ((1 << MCP7940_ALM0IF) | (1 << MCP7940_ALMPOL));         // Keep pol and alm flag bit        //
    wkdayRegister |= alarmType << 4;                                          // Set 3 bits from alarmType        //
    wkdayRegister |= (dt.dayOfTheWeek() & 0x07);                              // Set 3 bits for dow from date     //
```

Maybe it would also be beneficial to be able to set the polarity right in the setAlarm function with an additional optional paramter?

``` c++
bool setAlarm(const uint8_t alarmNumber, const uint8_t alarmType, const DateTime dt, const bool state = true, const uint8_t polarity = 0);
```

INemesisI, 2019-01-18:
Thank you very much!
After sleeping over it I noticed that adding the polarity to the setAlarm function makes no sense, since it would not work when using two alarms.
So you can ignore the last part!

SV-Zanshin, 2019-01-19:
Made changes and implemented

closed 2019-01-19 by SV-Zanshin