Zanduino/MCP7940 #41: MCP7940 how to set alarm to Day-of-the-week(1-7)

wimvanderneut, 2019-01-31:
Hello,

Can anybody help me te set an alarm at a day of the week(1-7). I want to create an interrupt(MFP) on a day-of-the-week, other than just a day in the month. is that possible? 

the line i am using is:

/*
* AlarmType Description
0 Alarms off
1 Minutes match
2 Hours match
3 Day-of-Week matches
4 Date matches
7 Seconds, Minutes, Hours, Day-of-Week, Date and Month match
*/

 MCP7940.setAlarm(0,3,DateTime(0,0,3,0,0,0));

kind regards,

Wim

SV-Zanshin, 2019-02-01:
Setting the Alarm to a day of the week works. You specified the mask as 3 in your example above, which denotes "day of week", but then you used the construct "DateTime(0,0,3,0,0,0)" I think you did something you didn't expect

```
  DateTime xxx(0, 0, 3, 0, 0, 0);
  Serial.print("Day of Week ");
  Serial.println(xxx.dayOfTheWeek());
```

returns day "1" Which I believe you probably didn't want. You should enter a valid date as input for the DateTime() function and the the MCP7940.setAlarm() will take the DOW from that date and set the alarm accordingly.

SV-Zanshin, 2019-02-01:
That means if you call `MCP7940.setAlarm(0,3,2019,02,01,0,0,0));` then it will set alarm 0 to trigger on a Friday.

wimvanderneut, 2019-02-01:
Hello, SV-Zanshin,

Your solution works!!! thank you very much!!

wimvanderneut, 2019-03-18:
Hello Zanshin,

Is it also possible to do a wake up after 10 or 30 seconds? How can i program that?

Kind regards,

Wim

SV-Zanshin, 2019-03-22:
Hello Wim,
You would need to read the time, add the seconds and then set the alarm.
-Arnd.

wimvanderneut, 2023-10-11:
Hello Zanshin,

is it possible to use the MCP7940 to interrupt the arduino mega by interrupt? and so yes, what am i doing wrong in the code below?
Kind regards,
 Wim

#include <MCP7940.h>  // Include the MCP7940 RTC library
/***************************************************************************************************
** Declare all program constants and enumerated types                                             **
***************************************************************************************************/
const uint32_t SERIAL_SPEED{9600};     ///< Set the baud rate for Serial I/O
const uint8_t  LED_PIN{13};              ///< Arduino built-in LED pin number
const uint8_t  SPRINTF_BUFFER_SIZE{32};  ///< Buffer size for sprintf()
const uint8_t  ALARM1_INTERVAL{15};      ///< Interval seconds for alarm
const byte interruptPin = 82;
boolean mfpPinTriggered = false;
/*! ///< Enumeration of MCP7940 alarm types */
enum alarmTypes {
  matchSeconds,
  matchMinutes,
  matchHours,
  matchDayOfWeek,
  matchDayOfMonth,
  Unused1,
  Unused2,
  matchAll,
  Unknown
};
/***************************************************************************************************
** Declare global variables and instantiate classes                                               **
***************************************************************************************************/
MCP7940_Class MCP7940;                           ///< Create instance of the MCP7940M
char          inputBuffer[SPRINTF_BUFFER_SIZE];  ///< Buffer for sprintf() / sscanf()

void rtcMFP() 
{
  mfpPinTriggered = true;
  Serial.println(F(" #*INTRTC*#"));//delay(50); 
}

void setup() {
  attachInterrupt(digitalPinToInterrupt(2), rtcMFP, RISING);// (recommended)
  //attachInterrupt(82, rtcMFP, RISING);//interrupt for rtc Alarm
  /*!
      @brief  Arduino method called once upon start or restart.
  */
  Serial.begin(SERIAL_SPEED);  // Start serial port at specified Baud rate
#ifdef __AVR_ATmega32U4__      // If this is a 32U4 processor, then wait 3 seconds for the serial
                               // interface to initialize
  delay(3000);
#endif
  Serial.print(F("\nStarting SetAlarms program\n"));
  Serial.print(F("- Compiled with c++ version "));
  Serial.print(F(__VERSION__));  // Show compiler information
  Serial.print(F("\n- On "));
  Serial.print(F(__DATE__));
  Serial.print(F(" at "));
  Serial.print(F(__TIME__));
  Serial.print(F("\n"));
  while (!MCP7940.begin())  // Loop until the RTC communications are established
  {
    Serial.println(F("Unable to find MCP7940. Checking again in 3s."));
    delay(3000);
  }  // of loop until device is located
  Serial.println(F("MCP7940 initialized."));
  while (!MCP7940.deviceStatus())  // Turn oscillator on if necessary
  {
    Serial.println(F("Oscillator is off, turning it on."));
    bool deviceStatus = MCP7940.deviceStart();  // Start oscillator and return state
    if (!deviceStatus)                          // If it didn't start
    {
      Serial.println(F("Oscillator did not start, trying again."));
      delay(1000);
    }  // of if-then oscillator didn't start
  }    // of while the oscillator is of

  Serial.println("Setting MCP7940M to date/time of library compile");
  MCP7940.adjust();  // Use compile date/time to set clock
  Serial.print("Date/Time set to ");
  DateTime now = MCP7940.now();  // get the current time
  // Use sprintf() to pretty print date/time with leading zeroes
  sprintf(inputBuffer, "%04d-%02d-%02d %02d:%02d:%02d", now.year(), now.month(), now.day(),
          now.hour(), now.minute(), now.second());
  Serial.println(inputBuffer);
  Serial.println("Setting alarm 0 for every minute at 0 seconds.");
  MCP7940.setAlarm(0, matchSeconds, now - TimeSpan(0, 0, 0, now.second()),
                   true);  // Match once a minute at 0 seconds
  Serial.print("Setting alarm 1 to go off at ");
  now = now + TimeSpan(0, 0, 0, ALARM1_INTERVAL);  // Add interval to current time
  sprintf(inputBuffer, "%04d-%02d-%02d %02d:%02d:%02d", now.year(), now.month(), now.day(),
          now.hour(), now.minute(), now.second());
  Serial.println(inputBuffer);
  MCP7940.setAlarm(1, matchAll, now, true);  // Set alarm to go off then
  pinMode(LED_PIN, OUTPUT);                  // Declare built-in LED as output
}  // of method setup()

void loop() {
  /*!
      @brief  Arduino method called after setup() which loops forever
  */
  static uint8_t secs;
  DateTime       now = MCP7940.now();  // get the current time
  if (secs != now.second())            // Output if seconds have changed
  {
    sprintf(inputBuffer, "%04d-%02d-%02d %02d:%02d:%02d", now.year(), now.month(), now.day(),
            now.hour(), now.minute(), now.second());
    //Serial.print(inputBuffer);
    secs = now.second();     // Set the counter for later comparision
    if (MCP7940.isAlarm(0))  // When alarm 0 is triggered
    {
      Serial.print(" *Alarm0*");Serial.println(inputBuffer);
      MCP7940.clearAlarm(0);
    }                        // of if Alarm 0 has been triggered
    if (MCP7940.isAlarm(1))  // When alarm 0 is triggered
    {
      Serial.print(" *Alarm1* resetting to go off next at ");
      now = now + TimeSpan(0, 0, 0, ALARM1_INTERVAL);  // Add interval to current time
      sprintf(inputBuffer, "%04d-%02d-%02d %02d:%02d:%02d", now.year(), now.month(), now.day(),
              now.hour(), now.minute(), now.second());
      Serial.println(inputBuffer);
      MCP7940.setAlarm(1, matchAll, now, true);  // Set alarm to go off in 10s again
    }                                            // of if Alarm 0 has been triggered
    //Serial.println();
  }  // of if the seconds have changed
}  // of method loop()

SV-Zanshin, 2023-10-11:
@wimvanderneut - yes, you can use the MCP7940 for interrupts. Check out the example example, [Square Wave](https://github.com/Zanduino/MCP7940/wiki/SquareWave.ino), to see see how a simple interrupt is used there. That can be adapted as you want.