Zanduino/MCP7940 #45: Battery backup not working if power to micro-controller is cut

UnexpectedMaker, 2019-04-04:
Or no clear example on how to do this properly.

BTW, thanks for making this library... it seems super thorough :-)

I have the TestBatteryBackup example working,  and I can disconnect 3V to the MCP7940N and then reconnect and it keeps time fine... but if I pull power to the microcontroller, and re-connect, time is reset.

I though it was due to the
`MCP7940.adjust(); `
Being called in start, but if I change that to look for the powerfail flag and and adjust based on the power restore time and clear the flag, it doesn't work.

Am I missing something here? The entire purpose of a battery backed RTC is to keep time when power to the main device is down, but I can't find any code setup that does this.

Thanks!
Seon
unexpectedmaker.com

UnexpectedMaker, 2019-04-04:
Ok, Ignore this, seems I don't need to do anything... no adjust at all... when the micro restarts  after pulling power, the time continues as normal! It was me thinking I needed to check the power flag myself and adjust time that was making it break!

Sorry for the noise... if there was a way to delete the issue I'd delete it ;)

Seon
unexpectedmaker.com

SV-Zanshin, 2019-04-04:
No problem at all. I'm glad you found the library useful and don't hesitate to ask any questions that you might have.

closed 2019-04-04 by SV-Zanshin

manaioussema, 2020-11-10:
hello guys,
In my case, i get the code stuck in "waiting for power" once i remove the VCC pin from 3.3V and get it back. I'm also attaching a CR2032 coin cell to VBAT pin.