Zanduino/MCP7940 #46: Cancel a setAlarm

logicaprogrammabile, 2019-12-12:
Hi
for particular project i need to cancel a setAlarm setting. For example i set a alarm, but in a specific condition is possible to cancel the alarm settings so that this is not generated?

Thanks

logicaprogrammabile, 2019-12-12:
i try this

`bool MCP7940_Class::resetAlarm(const uint8_t alarmNumber)
{
	clearRegisterBit(MCP7940_CONTROL, alarmNumber ? MCP7940_ALM1EN : MCP7940_ALM0EN); // Turn off the alarm
	return true;
}`

and work fine

SV-Zanshin, 2019-12-12:
Currently this is done using the SetAlarm() with an "AlarmType" of "0" and some value for the "AlarmDate" parameter which isn't used.

SV-Zanshin, 2019-12-13:
I don't want to add a new function to clear the alarm because the functionality already exists  in the SetAlarm() function. Is that acceptable to you? If so, I'll close this bug report.

logicaprogrammabile, 2019-12-13:
ok, i read the code and the function to reset alarm is setAlarmState(), not setAlarm. it's correct?

SV-Zanshin, 2019-12-13:
The https://github.com/SV-Zanshin/MCP7940/wiki/setAlarmState() just sets the state of the alarm, but doesn't enable/disable it, the https://github.com/SV-Zanshin/MCP7940/wiki/setAlarm() call with the "AlarmType" parameter set to 0 will disable it.

logicaprogrammabile, 2019-12-16:
i tested setAlarmState() and the function work fine. 
for example:
`MCP7940.setAlarmState(0, false);`
disable the alarm 0 correctly

SV-Zanshin, 2019-12-16:
Whether it works as you expect it to depends upon what type of alarm was triggered. If you set a "minutes match" type of alarm and turn it off using setAlarmState() but the minute is still the same then the alarm would immediately re-assert. If you set an alarm for a specific date/time then this method would work.

closed 2019-12-17 by logicaprogrammabile