Zanduino/MCP7940 #47: ATmega328PB & MiniCore incompatible??

wvmarle, 2020-01-22:
I've before worked with this RTC with the ATmega328P and it worked great. Now I've build a board around the new ATmega328PB, using the MiniCore, and I'm running into some weird problems.

First I found out the battery backup doesn't work. My multimeter shows voltage at the VBAT pin (about 2.5V - a fresh 3V battery minus a diode drop) , but it doesn't keep the time over power cycles as it should.

Also I had to replace the simple MCP7940.adjust() calls with this:

    char d[13];
    strcpy_P(d, PSTR(__DATE__));
    char t[13];
    strcpy_P(t, PSTR(__TIME__));
    DateTime dt = DateTime(d, t);
    MCP7940.adjust(dt);                                                         // Set to library compile Date/Time //

This code works properly setting the time to the compile time Jan 22 2020 at 22:30:04; the adjust() function sets the time to "2075-01-30 32:50:50" (yes, hour 32! No idea how that is even possible). This accounts for both the TestBatteryBackup sketch and the SetAndCalibrate sketch.

As long as the power is up, the time runs just fine. So the chip appears to be in proper working order, and communicates as it should. The ATmega328PB itself runs at 18.432 MHz, Vcc = 5V.

SV-Zanshin, 2020-01-23:
I will look into this, but my first thought is that it might be timing issues considering the clock is at 18.432MHz instead of the normal 16MHz. If the MiniCore doesn't account for this speed difference then this might cause communications problems. I don't have access to this chip, but will see if I can figure out what might be happening. I will be out of communications for a week (I'm sailing offshore from the USA to the Caribbean on a sailboat) but will then have some time available.

wvmarle, 2020-01-23:
It is mostly the __DATE__ and __TIME__ that go haywire; the moment I copy them and use that to call the adjust() function it works fine.

The clock speed shouldn't be the issue as I can select the frequency in the menus, the rest of the I2C works: reading the time goes fine. The ATmega328PB should be pretty much the same as the ATmega328P plus some really nice upgrades (such as a second UART and two more 16-bit timers)

The other issue is that the backup battery is seemingly not used. I did remember to enable it (wonder why the chip maker even thought that having to enable it explicitly is even a good idea...).

SV-Zanshin, 2020-05-18:
I will change the adjust() the way you did for your core, this will then work for any current or future core, as the compiler variables might be implemented in different ways.

Could it be that you have a MCP7940**M** chip - that doesn't support battery backup.

closed 2020-05-18 by SV-Zanshin