Zanduino/MCP7940 #49: hello guys,

SV-Zanshin, 2020-11-10:
hello guys,
In my case, i get the code stuck in "waiting for power" once i remove the VCC pin from 3.3V and get it back. I'm also attaching a CR2032 coin cell to VBAT pin.

_Originally posted by @manaioussema in https://github.com/SV-Zanshin/MCP7940/issues/45#issuecomment-724731016_

SV-Zanshin, 2020-11-10:
Do you remove VCC from 3.3V on MCP7940 or on the Arduino? Which Arduino are you using for this test?

manaioussema, 2020-11-10:
i removed VCC from 3.3V on MCP7940, i'm using a nucleo board.

SV-Zanshin, 2020-11-10:
And just to make sure, you are using a MCP7940N  breakout board or chip?

manaioussema, 2020-11-11:
i'm using a breakout board that i made (just the crystal and exposing the pins) and yes i'm using the "testbatterybackup.ino"

SV-Zanshin, 2020-11-11:
OK, let me see if I can find a MCP7940N in my box here and replicate the issue.  Do you have pull-ups on the I2C lines? And could you post the output of the sketch to this issue?

manaioussema, 2020-11-11:
Yes i've the pull-ups on the I2C lines.
[image]

SV-Zanshin, 2020-11-11:
I can't find an MCP7940N in my box, all I have are MCP7940M here.

I do know that I have tested it and have a running solution elsewhere.  Are you certain that you hooked up the CR2032 to Vbat and Vss and that the coin cell has more than 1.3V? If the battery backup power were not present when the main power fails, then the system would wait forever (the way it is happening now).

manaioussema, 2020-11-11:
i'm pretty sure that the coin cell has more than 3V, it's brand new. And yes, all the wiring is ok

SV-Zanshin, 2020-11-11:
Can you put a line from the 3.3V power supply on the microprocessor to VBAT instead of using the battery and then just disconnect the Vcc line to see if the problem remains the same.

Basically if the power to Vcc goes down and the battery backup has no power or is not enabled, then when the MCP7940 gets power on Vcc again it won't run, it needs a call to "MCP7940.deviceStart()" to start the oscillator. So if you hit the RESET on your board after the sketch hangs at "Waiting for power..." and the clock runs again, it points to either of those two conditions.

manaioussema, 2020-11-11:
i tried to put VBAT pin on 3.3V of the µC and i get the same result

SV-Zanshin, 2020-11-11:
Can you change the 
`while(MCP7940.now().second()==secs && (millis()-startMillis<10000);`

with 
`long startMillis = millis();
    while(MCP7940.now().second()==secs && (millis()-startMillis<10000);
	`

and re-apply the power within 10 seconds and paste the output?

manaioussema, 2020-11-11:
there is no such line in the example "testbatterybackup.ino"

SV-Zanshin, 2020-11-11:
Line 98

SV-Zanshin, 2020-11-11:
I see, my code quotes were in the wrong place.
I'll repeat:
Replace line 98 = `while(MCP7940.now().second()==sec)s`
with
`	long startMillis = millis();
    while(MCP7940.now().second()==secs && (millis()-startMillis<10000);
`

manaioussema, 2020-11-12:
this is what i get after the modification of the code, the last one appeared even after i put back the VCC.
[image]

SV-Zanshin, 2020-11-14:
I rummaged around my boxes this morning and found a board where I'd soldered a MCP7940N and when using an UNO rather than a Micro I can now reproduce the error.  I will look into this further today and tomorrow and will get back on this bug.

SV-Zanshin, 2020-11-15:
Fixed issue and made it part of a new version, v1.1.8

closed 2020-11-15 by SV-Zanshin