Zanduino/MCP7940 #60: DateTime(0) doesn't resolve correctly to the epoch

obdevel, 2021-05-19:
## Expected Behavior

A DateTime object initialised to zero should resolve to the unix epoch, i.e. 1/1/1970 00:00:00

I'm trying to initialise the time component to 00:00:00. This seems the easiest way.

## Actual Behavior

The actual output is 6/2/2106 6:28:16

## Steps to Reproduce the Problem

Minimum reproducible code:

```
#include <MCP7940.h>

void setup() {
  Serial.begin(115200);

  DateTime dt1 = DateTime(0);
  // DateTime dt1(0); //  produces same result
  Serial.println(dt1.unixtime());
  Serial.print(dt1.day());
  Serial.print("/");
  Serial.print(dt1.month());
  Serial.print("/");
  Serial.print(dt1.year());
  Serial.print(" ");
  Serial.print(dt1.hour());
  Serial.print(":");
  Serial.print(dt1.minute());
  Serial.print(":");
  Serial.print(dt1.second());
  Serial.println();
}

void loop() {
}
```
Output is:
```
0
6/2/2106 6:28:16
```

If I construct the object as
```
DateTime dt1(1970, 1, 1, 0, 0, 0);
```
the output is:

```
2269013504
1/1/2178 0:0:0
```

## Specifications

  - Library Version: 1.2.0
  - IDE Version: 1.8.13
  - Platform: Arduino Uno R3

SV-Zanshin, 2021-05-20:
That certainly looks like bug! I'm not sure I need to dig out a MCP7940 to debug this issue, it looks like the issue is in the time classes and not the hardware.

SV-Zanshin, 2021-05-23:
The library uses a fork of Jeelabs RTC, which is not maintained by a fork at https://github.com/adafruit/RTClib 

This DateTime class is limited to dates starting at 2000-01-01. Passing a UnixTime less than that date causes library underflow errors which explains the output you are seeing. I've changed the call to DateTime() to not allow any date lower than that value.

Changing the library to allow for the full UnixTime range would mean increasing data types everywhere and adding logic. Considering that the MCP7940 can't deal with dates before 2000 the library doesn't need changing.

I will update the wiki as well.

obdevel, 2021-05-23:
Thanks Arnd. I don't need dates/times before 2000. I guess I was led astray by the naming of the unixtime() method and assumed it used the Unix epoch like time_t based libraries. 

I'll let you close this issue once you've made the update.

SV-Zanshin, 2021-05-23:
The library does use UnixTime starting at the correct epoch date, it just doesn't any values less than the number of seconds defined at:

`const uint32_t SECS_1970_TO_2000{946684800};   ///< Seconds between year 1970 and 2000`

closed 2021-05-23 by SV-Zanshin