Zanduino/MCP7940 #63: Working with ESP chips

BrotherV, 2021-12-16:
**Adding new begin() method for selectable pins of Wire library**
_To communicate ESP chips to MCP7940, we need to set up pins, which means we need to pass the number of GPIO to the MCP7940 class_

**Adding a new begin(SDA_PIN, SCL_PIN) method to the library**
_To connect the chips, in which i2C pins can be selected, a new method should be added to the library in both .h and .cpp files_

**Adding begin(SDA_PIN, SCL_PIN) header file**
`bool begin(byte i2cPin1, byte i2cPin2te, const uint32_t i2cSpeed = I2C_STANDARD_MODE) const;`

I have changed and tested your library with ESP chips and have attached them here.
[MCP7940.zip](https://github.com/Zanduino/MCP7940/files/7724914/MCP7940.zip)

SV-Zanshin, 2021-12-16:
I see what you mean, even for other devices user might want an alternative SDA and SCL pin.  I can add an overloaded begin() to define these.

You attached a .zip file which is unreadable. Perhaps you could just add the code you wrote for the begin() method and I'll add in the rest.

BrotherV, 2021-12-16:
Thanks for your consideration. I have added codes here.

**Adding begin(SDA_PIN, SCL_PIN) to MCP7940.h file**
```ruby
    bool begin(int sda, int scl, const uint32_t i2cSpeed = I2C_STANDARD_MODE) const;
```

**Adding begin(SDA_PIN, SCL_PIN) method to MCP7940.cpp file**
```ruby
    bool MCP7940_Class::begin(int sda, int scl, const uint32_t i2cSpeed) const
    {
      /*!
          @brief     Start I2C device communications
          @details   Starts I2C comms with the device, using a default speed if one is not specified
          @param[in] i2cSpeed defaults to I2C_STANDARD_MODE, otherwise use speed in Herz
          @return    true if successfully started communication, otherwise false
      */
      Wire.begin(sda, scl);                    // Start I2C as master device
      Wire.setClock(i2cSpeed);                 // Set the I2C bus speed
      Wire.beginTransmission(MCP7940_ADDRESS); // Address the MCP7940
      if (Wire.endTransmission() == 0)         // If there a device present
      {
        clearRegisterBit(MCP7940_RTCHOUR, MCP7940_12_24); // Use 24 hour clock
        setRegisterBit(MCP7940_CONTROL, MCP7940_ALMPOL);  // assert alarm low, default high
        return true;                                      // return success
      }
      else
      {
        return false; // return error if no device found
      }               // of if-then-else device detected
    } // of method begin()
```

SV-Zanshin, 2021-12-16:
I've solved it a bit differently so that the begin() functions have only one code base. I think that what I've just uploaded to the main branch should work for the ESP8266 as well as for any other devices and also be backwards compatible. Can you check the changes to the library .c and .h files?

SV-Zanshin, 2021-12-18:
clang-format is quite finicky! But I've got a correct version on the master branch for you to try out. It compiles correctly on all platforms.

closed 2025-01-26 by SV-Zanshin