Zanduino/MCP7940 #70: Exactly one alarm per day ?

Greece2, 2025-01-24:
Hi, tried to figure out how to set the alarm in way that the MFP ist triggered exactly once every 24 hours. So far - no success :-(
Any idea how that could be done ?

Thanks already,  Greece2

SV-Zanshin, 2025-01-26:
The [SetAlarm()](https://github.com/Zanduino/MCP7940/wiki/setAlarm()) function allows you to do that. For example, setting

MCP7940.setAlarm(1,2,DateTime(2017,8,5,18,30,0));

Will trigger every time the hour is 18 (6PM)

closed 2025-01-26 by SV-Zanshin

Greece2, 2025-01-27:
That's great !

I just try to figure out the parameters
(1,2, DateTime(2017,8,5,18,30,0))
1 = ?
2 = ?
2017 -> year (not relevant)
8 -> could be month (not relevant)
5 ->could be day (not relevant)
18 = hours
30 = ?
0 = minutes?  seconds ?

Thanks for your help !

Best regards, Greece2

SV-Zanshin, 2025-01-27:
The functions are all documented here on git, if you click on the hotlink in my first response it takes you to the documentation page which describes all the parameters to the function, and click on [DateTime()](https://github.com/Zanduino/MCP7940/wiki/DateTimeClass) for the description of the DateTime function

SV-Zanshin, 2025-01-27:
So "MCP7940.setAlarm(1,2,DateTime(2017,8,5,18,30,0));" is>

Set Alarm number 1
use hours match
Match August 5 2017 at 18:30

Since the alarm is set to match only the hour, the rest of the date is ignored. So it will trigger every time 18:00 rolls around.

Greece2, 2025-01-28:
Hi, thanks for explaining !
The 18:30 setting versus "only match hours" so the alarm will go off at 18:00 did confuse me.
But very well explained. And thanks for pointing out the documentation, that's very helpful indeed.

Best regards, Greece2

SV-Zanshin, 2025-01-28:
No worries, I'm glad you are using the library and find it helpful.